Lenti84/VdMot_Controller #1: Great Project

xal88, 2021-04-09:
Hallo, 
habe  von deinem  Projekt im FHEM Forum gelesen.
Das wäre genau, was ich mir für meine Deckenklimatisierung als Steuerung wünsche. 
Die ist nicht so träge wie eine Fußbodenheizung und braucht ordentliche Regelung, nicht nur 0/1.
Daten der Wärmepumpe, VL/RL und Thermostate habe ich schon in MQTT.
I2C fur HTU21D ( Humidity/Temperature -> Taupunkt im Verteiler )  würd ich am ESP dazubauen... ( die Onewire DHT22 sind Mist bei 100% Feuchte )...
Wenn du Prototypen/Kleinserie machst,  beteilige ich mich gerne an den Kosten und würd 1 zum testen  und ggf. 3 Stk für Echt nehmen
danke+lg
Alf

Lenti84, 2022-01-04:
Info: hinsichtlich Bausatz bitte immer mal wieder im FHEM Forum vorbeisehen: https://forum.fhem.de/index.php/topic,103699.0.html

xal88, 2022-01-07:
Hallo, 
wegen Bausatz hab ich dir eine PM im FHEM Forum gesendt.
Wie sieht es aus mit feature requests für die nächste Serie? Mich würde ein Anschluss für einen Durchflussmesser ( interrupt/Hall ) interessieren, weiterverarbeitung dann über MQTT :-)
lg Alf

Lenti84, 2022-01-11:
Ziel für die nächste Revision ist quasi nichts zu ändern. Kann dir deshalb leider nichts versprechen.

Im aktuellen Release gibt es aber sogar noch ungenutzte Pins, die auf eine Stiftleiste herausgeführt sind.
Über X15:Pin7 (INT) lässt sich sicher ein externer Schalter anschließen. PullUp wäre auch vorhanden.
Softwareseitig müsste geklärt werden, ob der Pin PB8 für externe Interrupts genutzt werden kann.

Beste Grüße
Andreas

xal88, 2022-01-12:
Danke, alles klar. Ich glaub da nehm ich lieber einen IO vom ESP32 - ist auch einfacher für die SW-Erweiterung.
Mit dem STM32 kenn ich mich nicht aus...

closed 2022-01-12 by xal88