Lenti84/VdMot_Controller #2: C-sample Hardware Issues and Change Requests

Lenti84, 2021-09-06:
I'm planning to design an updated version of the hardware. Next step of hardware will be called C-sample.
Therefore i want to collect known issues and possible change requests.

**currently known issues**
- B4: footprint for dc-dc converter: crossed secondary side pinout for gnd and vout
- V3: false assembly of NPN bipolar instead of PNP type
- X15, 21: hole diameter to small
- X1...X12: connection to MEBP_4-4S wrong: interchange pin 1 with pin 2
- no holes for case screws

**change requests**
- current consumption of L293 is 50 mA each, should be lowered, maybe U1, U3, U8 should be shut down during idle
- voltage devider at net "Sense" to match voltage levels better from opamp to STM32
- bypass capacitor for ESP8266
- replace relay K1 by analog switch
- support STM32F401 blackpill boards instead of STM32F103 bluepill
- maybe switch from ESP8266 to ESP32 or WT32-ETH01
- ....

Please feel free to comment your suggestions so maybe i can consider implementing them.

nstadigs, 2021-10-12:
Hi! Thanks for this great project!

I have 13 valves to control. Would it be possible to squeeze in control for an extra connector (or two, to make the number even)? Or maybe add support for an extension board? :grimacing: 
I know this might be a big request, but I had to ask :)

Lenti84, 2022-01-04:
c-sample revision of hardware implemented --> closed

closed 2022-01-04 by Lenti84