Lenti84/VdMot_Controller #9: Why are there 2 MCUs?

unaiur, 2021-12-31:
Great project! It is something I has need searching for a while but I have one question... why is the second MCU required?

ESP32 is quite capable, has 2 "high" speed cores plus a low power coprocessor and plenty of peripherals, like a Hall effect sensor and an   opamp to measure current and a ADC that is fast enough to count revolutions by software.

I am sure you have some reason to use the STM mcu as the core instead. I just would like to know.

Thank you very much!

Lenti84, 2022-01-04:
I wanted to make the VdMot Controller a kind of modular.
STM32 is the core of the system for motor control, valve managing, RS485, 1-wire and optional CAN.
Wifi / Ethernet is only an option and realised via the ESP32 (older hw revision ESP8266).

If you look the other way round the STM32 is a cheap and intelligent port extension for the ESP ;-)
Meanwhile the STM32 it is fully flashable via ESP32 web interface...

unaiur, 2022-01-06:
Ummm, I would like to use the low power RiscV core inside Esp32s2 instead, to reduce the component count and make it simpler (I am a software guy).

Also, I will try to use the 13bit/6Khz ADC in that core to measure the active motor current across a 1ohm resistance and try to identify the back EMF using software implemented low/high pass filters.

xal88, 2022-01-07:
> Ummm, I would like to use the low power RiscV core inside Esp32s2 instead, to reduce the component count and make it simpler (I am a software guy).
> 
> Also, I will try to use the 13bit/6Khz ADC in that core to measure the active motor current across a 1ohm resistance and try to identify the back EMF using software implemented low/high pass filters.
> 
> Do you know the maximum hz of the motors in your project? If they run at more that 15.000 rpm it won't be possible to implement it by software, but seems unlikely given the low voltage they use.

Hi, 
I think that the project has a good modular HW design for now, with the ESP32 a very common, good available cheap chip.
And you can flash the STM32 from webserver on ESP...
The ESP2S has only 2 13 bit ADCs, so this is no option. And it is a single Core CPU, and the old ESP single cores had problems with interrupts when using WiFi....
But in the end its Andreas design, and he has put a lot of effort into the current design.

Lenti84, 2022-01-11:
I think to remember the motors do approx 8000 counted revolutions in 60s running time. So this should be slow enough for a software appraoch then.

Please keep in mind that with actual hardware concept you will need more pins than the usual ESP32 can offer. Thus you will need a port expander oder something similar.

But at the actual state of the project i cant justify such fundamental changes...

unaiur, 2022-01-12:
The Esp32s2 chip has two cores; a fast core and an ultra low power core
that is always on. Nonetheless, the ultra low power core has more than
enough power to control the motors.

I only plan to drive one motor at a time. I will select the active L298
chip using 8 pins and all 8 L298 will share the same 4 input lines. I will
use another pin to read the motor current: 13 pins in total to manage 16
motors.

The Esp32s2 has more than 20 available pins, so I have plenty of IO
available in case I need an external ADC or the hardware spin counter you
have.

I will keep you updated about my experiment.
I already ordered the pieces. If it works, I would try to build a single
board with all the components instead of 10 different interconnected boards.

Thank you very much!

Lenti84, 2022-01-26:
I wish you success! Keep us updated.

closed 2022-01-26 by Lenti84