Lenti84/VdMot_Controller #37: 1-Wire not seeing all SDS18B20

xal88, 2022-06-23:
Some sensors are not detected by scan, others detected but no data.
GUI/config scan gives different results to the scans when booting.
Some sensor delivers data, if manually eintered ID in GUI/config

All sensors are tested and working with ESP PIN and 4k7 Ohm resistor. 
Sensor answer is on the bus, but not picked up to i2c?

### Signaling on oszilloscope comparing ESP and VdMot looks ok.
[image]

[image]

### Logic compare "-22"  sensor detected, but no data,
ESP test sample OK
[image]
VdMot
[image]

### Logic compare "a9" sensor not detected
ESP Test sample OK
[image]
VdMot test
[image]

Lenti84, 2022-06-25:
If you compare the oscilloscope shots from ESP and STM you can recognize, that the voltage level with STM32 is not perfectly correct. 
This is a flaw in the STM32 version of arduino 1-wire lib. But for my config it is working fine for currently 10 sensors.

I have tested a fix some time ago. I think it was in OneWire_direct_gpio.h
Instead of a PushPull output
`#define DIRECT_MODE_OUTPUT(base, pin)   pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_OUTPUT_PP, GPIO_NOPULL, 0))`
a change to OpenDrain changed the voltage level problem
`#define DIRECT_MODE_OUTPUT(base, pin)   pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_OUTPUT_OD, GPIO_NOPULL, 0))`

Maybe give it a try if this changes something.
I think i will integrate that in one of the next releases...

SurfGargano, 2022-11-12:
I have checked with 15 sensors and the modification in V1.05 of stm software: changing from push-pull to open drain.
It works.

xal88, 2022-11-12:
Thanks. Tried to compile the stm-1.05 with VSC/platformio, result is only 52 kB, old one was 198kB.
With esp32_1.02 and esp32_1.04 the file is growing from  1012->1128 kB.
Can this be right, could you maybe upload the stm firmware 1.05?

SurfGargano, 2022-11-13:
yes, 52 kB for stm 1.0.5 is correct The old one was about 48 kB
the esp file is 1106 kB with V1.0.5
I have uploaded the stm bin files in the bin directory in developer branch in my repo.
https://github.com/SurfGargano/VdMot_Controller/tree/developer/software_stm32/bin/v1.0.5

xal88, 2022-11-13:
WTH32 is at 1.05. OTA worked.
Tried to update the STM32 from 1.02 to 1.05.  
Upload of the file ok, but after selecting file, and clicking "Update STM32 now" the progress bar goes to 5%, and then "error". 
Self comiled firmware.bin or development branch same issue.
What am I doing wrong?

SurfGargano, 2022-11-13:
Pls try following : Format the spiffs by clicking the Format Button in Config section of the wth32. Upload stm bin file again and start the update of stm.

xal88, 2022-11-14:
Tried on 2 different VdMot boards, firefox or chrome, with WT32 on 1.05.
"/config/Format FS" gives no status update, just reboot. Uploaded Files are gone afterwards.
"update/ST32" gives 5%, than error and reboot.
Syslog atomic shows no useful messages, only the ones about the valve positions.
What else can I do - Blackpill update via external serial?

SurfGargano, 2022-11-14:
After Format the file info on the config page should be invisible. Is the stm still working?What does the serial debug of wt32 says?

Lenti84, 2022-11-14:
Have you tried initial flash with Pressung boot0 button in BlackPill Board? 

https://github.com/Lenti84/VdMot_Controller/blob/master/assembly_kit/installation.md#initial-stm32-flash-via-vdmot-controller

xal88, 2022-11-14:
update to 1.05_Dev initial flash did work, thanks for your help.
Connected 32 senors ( all I have ), found all - even the cheap chinese problematic ones.
Will close the issue.

closed 2022-11-14 by xal88