Lenti84/VdMot_Controller #45: Multiple controllers for one room

zillawilli, 2022-11-11:
Hello

I have the following problem.

In my case, the kitchen is quite large, so the installer has installed 4 loops of floor heating.
Two are short, as these are to heat only a narrow edge to the outside wall.
The other two loops are long and heat the rest of the kitchen.
Each loop has its own valve.

As I understand it, only the control of one heating circuit per room is possible, right?

Would it be possible to group several circuits of a room together?

I would imagine this to be the case:
The inlet temperature must be measured with a single sensor.
Now, in order to theoretically determine the individual lengths of the combined heating circuits, a reference measurement must be made.

For this purpose, the temperature and the time until a certain temperature is reached is measured at the return of each circuit. The longest heating circuit requires the most time, the shortest accordingly less.
Based on this determined time, the longest heating circuit is now determined to the master and its return temperature is used as the setpoint for the other circuits. At the master, for example, the valve is 100% open, the other circuits correspondingly less.
If now the master must be opened more than 100%, a message is needed because of too low inlet temperature.
If now the valve with the smallest opening would have to be closed by the control, a message with too high inlet temperature is needed.

What do you say to this? Would something like this be possible?

greetings
William

SurfGargano, 2022-11-11:
If you have several loops of floor heating in one room , you have to make a hydraulic adjustment. That is important.

You can group the valves together in the control section, if you use the on-board valve controller.
But only one thermometer is used.

I have 3 loops of floor heating in living area, dining area and kitchen in one room. It works.

[image]

xal88, 2022-11-11:
Hi,
VdMot is flexible with assigning sensors to multiple circuits. And your control strategy is completely in your hands.
Without knowing the general setup ( what heat source, pump type and flow... ) its difficult to advise.
I would go with a static machanical hydraulic balance setup for your 4 kitchen valves, no need for motorvalves there.
Treat them as one circuit and control via heating system
BR Alfred

zillawilli, 2022-11-15:
Hello

Thank you for your responses, which have inspired me to think further.

The procedure for rooms with multiple heating circuits, in my case with four circuits, would now be as follows:

I measure the return temperature over time for all circuits.
Now all valves are opened completely and the flow reducers are also set to 100%.
After a time of 30 minutes the log files are read out and examined in a program.
Depending on the length of the heating circuit, a certain return temperature will be set.
The circuit with the longest pipe has the lowest temperature. The shortest has the highest.
Now the flows are reduced until all have the same return temperature as the longest heating circuit. But the valves are all still 100% open.
That is what I understand by hydraulic balancing.
Only now each valve is controlled individually, so that the return temperatures are all the same.

kind regards
William

Hallo

Danke für eure Antworten, die mich zum weiteren Nachdenken angeregt haben.

Der Ablauf bei Räumen mit mehreren Heizkreisen, in meinem Fall mit vier Kreisen, wäre nun folgend:

Ich messe bei allen Kreisen die Rücklauftemperatur über die Zeit.
Nun werden alle Ventile komplett geöffnet und die Durchflussminderer ebenso auf 100% gestellt.
Nach Ablauf einer Zeit von 30min werden die Logdateien ausgelesen und in einem Programm begutachtet.
Je nach Länge des Heizkreises wird sich eine bestimmte Rücklauftemperatur einstellen.
Der Kreis mit dem längsten Rohr hat die niedrigste Temperatur. Der kürzeste die höchste.
Nun werden die Durchflüsse so lange reduziert, bis alle die gleiche Rücklauftemperatur wie der längste Heizkreis aufweisen. Die Ventile sind aber alle noch zu 100% geöffnet.
Das ist, was ich unter einem hydraulischen Abgleich verstehe.
Nun erst wird jedes Ventil einzeln angesteuert, so daß die Rücklauftemperaturen alle gleich sind.

freundliche Grüße
William

Lenti84, 2023-03-20:
Moved information to: https://github.com/Lenti84/VdMot_Controller/wiki/FAQ

closed 2023-03-20 by Lenti84