Lenti84/VdMot_Controller #53: ESP32 Software Ver. 1.05 no longer accepts changes.

SteffenB71, 2022-12-26:
ESP32 Software Ver. 1.05 no longer accepts changes. I can no longer add valves or change the station name. The old settings are always there again after save config.
Already deleted the ESP and flashed it again, but still no change. Remedy has version 1.02 because everything is normal.
Got the C2 board.

Steffen

Lenti84, 2022-12-29:
Had a look at it. With V1.05 the setting the station name is fine. Also detecting valves and naming them.
I switched between 1.02 and 1.05. Both seem to work correctly.

But i discovered a flaw when using some more 1-wire sensors.
Linking to the valves is only possible for some of the first valves. The last ones will never be linked to any 1-wire sensor.

A fix will be available soon. Had to rework the communication between STM32 and ESP32 a bit. Also reset behaviour.

Maybe you can check the behaviour with next release, as there are some other changes regarding config of controller.

SteffenB71, 2022-12-30:
Hello
I've even flashed a new ESP now, but it still appears the same. It does not save any new settings. Where are all the settings saved, that even a new ESP has the same Venoman.

Steffen

Lenti84, 2022-12-30:
Hmm, most of the settings are saved into the flash of the ESP32/WT32. For example the station name is stored there. But 1-wire settings are saved into the serial EEPROM connected to the STM32.

Does in your setup some settings exist, which will be stored correctly? Can you give a short list?

At the moment I have not really an idea. But i will look for a solution...
Maybe using a newer firmware version will help. I will soon publish version 1.07. Please try with this version and report.

best regards

Lenti84, 2023-01-04:
Please try newest firmware: https://github.com/Lenti84/VdMot_Controller/releases/tag/v1.0.7

SteffenB71, 2023-01-05:
Hi
With version 1.07 everything can be set and saved again. Thanks for the support. I have to update the STM chip as in the instructions or because the ESP can be updated online without having to go to the circuit board.
Steffen

Lenti84, 2023-01-07:
Dont know if I understand right.
But you can flash both Controllers via the web interface. No need to go to the circuit board.
For STM32 just make sure X20 is jumpered and you have selected "normal Partition".

Glad to hear it ist working again.

Best regards

SteffenB71, 2023-01-07:
Hi
That was exactly my question. When flashing the STM32 I have to set the jumper X20 and press the Boot0 button after the start:
1. Short jumper X20 - this allows the ESP32 to Reset STM32 and start the Bootload sequence
2. Power up VdMot Controller
3. Open VdMot Controller website and goto "Update"
4. Click on "Update STM32"
5. Choose the STM32 firmware file to upload
6. Click "Upload STM32 file"
7. Choose Option "boot partition" for initial flashing
8. After Upload choose correct file in filelist
9. Press BOOT0 Button at BlackPill Board until flashing progress is >15%
10. Click "Upload STM32 now"

Now write I should set to "normal partition". So you set the "Boot0 partition" only the very first time.
thanks

Lenti84, 2023-01-30:
Updated https://github.com/Lenti84/VdMot_Controller/blob/master/assembly_kit/installation.md.
Now there is distinction between regular and inital flashing of STM32.

closed 2023-01-30 by Lenti84