Lenti84/VdMot_Controller #54: problems with the mqtt conversation

Dobi62, 2023-01-01:
Hello
Between Christmas and New Years I soldered my board together and installed the latest software. 
Mqtt is running and I get all values ​​in FHEM but can't send any. When I send values, no matter what, I see them being sent, but the values ​​do not arrive in the VdMot. This applies to sending from FHEM as well as from the MQTT Explorer. 
These are similar errors as xal88 had in june 22.
What are the temp1 and temp2 fields in the Status tab for?
Are these placeholders for should and actual value?
How can I set this?
Is there anything to consider? Is there an explanation of the software on the internet?
I am thankful for every help.

Lenti84, 2023-01-02:
Please have a look at the software documentation at: https://github.com/Lenti84/VdMot_Controller/blob/developer/software.md
Please check, if setting the checkbox "Target" under "Protocoll/Publish" makes any difference.

For further debugging you can use a Syslog server installed to your PC. I'm using "Visual Syslog Server 1.6.3".
With a syslog server you can view a lot of debugging info of the controller.
For example a received MQTT message for a valve target would be showen as: "_Message: found topic WZSued/target : 16_"
Please setup the syslog server at the config tab, section "Network", at the bottom.

And here is an example piece of code how i have set up my controller under FHEM:
```
define mqtt MQTT 127.0.0.1:1883

define VdMot_Valve_1 MQTT_DEVICE
attr VdMot_Valve_1 IODev mqtt
attr VdMot_Valve_1 devStateStyle style=color:blue;;font-weight:bold;;text-align:right
attr VdMot_Valve_1 room MQTT
attr VdMot_Valve_1 stateFormat Ist: actual Status: state Strom: meancurrent Temp: temp1 Temp2: temp2
attr VdMot_Valve_1 subscribeReading_actual /VdMotFBH/valves/TestV/actual
attr VdMot_Valve_1 subscribeReading_meancurrent /VdMotFBH/valves/TestV/meancur
attr VdMot_Valve_1 subscribeReading_state /VdMotFBH/valves/TestV/state
attr VdMot_Valve_1 subscribeReading_temp1 /VdMotFBH/valves/TestV/temp1
attr VdMot_Valve_1 subscribeReading_temp2 /VdMotFBH/valves/TestV/temp2

define Heizung_Bad_Slider2 dummy
attr Heizung_Bad_Slider2 room MQTT
attr Heizung_Bad_Slider2 setList state:slider,0.0,1.0,100.0
attr Heizung_Bad_Slider2 webCmd state

define Heizung_Bad_SliderNotify2 notify Heizung_Bad_Slider2:* { \
my $sliderwert = \
ReadingsVal("Heizung_Bad_Slider2","state","on");; \
fhem ("set mqtt publish /VdMotFBH/valves/TestV/target $sliderwert");;
}
```

Hope this helps to find the problem.

Dobi62, 2023-01-02:
Thanks for the answer. 
That with the syslog definitely has to be stored somewhere. Also thanks for the fhem definitions, but I switched everything to mqtt2 about 1.5 years ago. But that's not the problem either. 
After I set everything to zero, named everything again and then rebooted the system, I can at least open and close the valves. The whole thing even though the topic target is not transferred at all. It is not possible to enter temp1 and temp2.
They stay at -50. 

That would be enough for regulation using Fhem. But now, in the heating season, it wouldn't be bad if VdMot could quickly take over control. 
Somehow I also have in my head to have read that this is possible.
But therefor you have to be able to transfer the target value and actual value. But which values ​​do I have to set and how?
I cannot set temp1 and temp2.

Set MQTT2_FHEM_Server public VdMot/valves/Kueche/target 65  works and sets the kitchen valve to 65%
Set MQTT2_FHEM_Server public VdMot/valves/Kueche/temp1 22 doesn't work and no reaction in the syslog.

anyone have any ideas

I do not anymore

Lenti84, 2023-01-02:
Temp1 and temp2 can't be set via MQTT. They are only published to broker according to the setup on config page for every valve.
To every of the 12 valves, max. 2 of the 1-wire sensors can be linked to. For example for measuring return flow temperature or floor temperature. The temperature of this 2 sensors will then be published via MQTT under the subtopic of the valve. So you do not have to worry about which 1-wire sensors belongs to which valve when controlling over MQTT.

If you want to use the integrated controller (tab "control"), you can set the actual and target value by MQTT via following topics  as documented under: https://github.com/Lenti84/VdMot_Controller/blob/developer/software.md, section _MQTT Interface WTH32_

topic | direction | description
-- | -- | --
maintopic/valves/valvesubtopic/tValue | subscription | actual temperature value for integrated valve position control
maintopic/valves/valvesubtopic/tTarget | subscription | temperature setpoint for integrated valve position control

With this MQTT values the integrated PI controller will calculate the needed valve position.

Is this what you want to do?

SurfGargano, 2023-01-04:
Using fhem please uncheck the 'Target' checkbox in config/protocol/publish

[image]

Dobi62, 2023-01-04:
Thanks for all the info.
As I said, closing and opening the valves now works wonderfully. Not with setting an actual temperature and the target temperature. They should appear under Control/Value or Control/Target. Or?
So I'll put the whole thing aside for now and take care of it again in the summer. In any case, there is no time to try it out during the heating period.
It would be nice if a few examples were described somewhere.

closed 2023-04-28 by SurfGargano