Lenti84/VdMot_Controller #55: ESP32 with Lan

Elluminatus, 2023-01-03:
Hi I‘m wondering if there could be a esp32 with Lan possible on the board? Thank you for your help

Greetings Thomas

Elluminatus, 2023-01-03:
Oh I saw that the C Sample has a Lan Port available

SurfGargano, 2023-01-04:
The esp32 we used is the WT32 - ETH01. Thie module has wifi and eth on board.

Elluminatus, 2023-01-04:
Thank you very much. I send you some questions via Tindie, maybe you can contact me.

 

Gesendet: Mittwoch, 4. Januar 2023 15:56
An: Lenti84/VdMot_Controller ***@***.***>
Cc: Elluminatus ***@***.***>; Author ***@***.***>
Betreff: Re: [Lenti84/VdMot_Controller] ESP32 with Lan (Issue #55)

 

The esp32 we used is the WT32 - ETH01. Thie module has wifi and eth on board.

Lenti84, 2023-01-12:
Your welcome. See my answers at Tindie. Kind regards.

closed 2023-01-30 by Lenti84