Lenti84/VdMot_Controller #59: Flash ESP32

bluephoenix28, 2023-01-30:
Hi,
I am running Linux and want to flash the ESP32 with the esptool. To what adresses do I need to flash the precompiled binaries bootloader_dio_40m.bin and boot_app0.bin?

Lenti84, 2023-01-30:
Did you have a look at https://github.com/Lenti84/VdMot_Controller/blob/master/assembly_kit/installation.md ?
See section _Initial ESP32 / WT32 Flash via Flash Download Tool_.

bluephoenix28, 2023-01-30:
Sorry, I missed that out. Now things are running, thx. 
For those who are also using Linux, I paste here how to flash the black pill and how to flash the ESP32.

---------------------------------------------------------------------------------
Flash the black pill with dfu-util, not connected to the Vdmot board

1) Remove the USB cable
2) Press and Hold the buttons NRST and BOOT0
3) Connect the USB-C cable
4) Release the NRST button
5) Wait 1 second
6) Release the BOOT0

Check that the device is found by Linux
$ dmesg

[ 4416.393784] usb 1-4: new full-speed USB device number 12 using xhci_hcd
[ 4416.646928] usb 1-4: New USB device found, idVendor=0483, idProduct=df11, bcdDevice=22.00
[ 4416.646930] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4416.646930] usb 1-4: Product: STM32  BOOTLOADER
[ 4416.646931] usb 1-4: Manufacturer: STMicroelectronics
[ 4416.646932] usb 1-4: SerialNumber: 3454367C3430

Check that the device is found by dfu-util
$ sudo dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [0483:df11] ver=2200, devnum=30, cfg=1, intf=0, path="1-4", alt=3, name="@Device Feature/0xFFFF0000/01*004 e", serial="348236533430"
Found DFU: [0483:df11] ver=2200, devnum=30, cfg=1, intf=0, path="1-4", alt=2, name="@OTP Memory /0x1FFF7800/01*512 e,01*016 e", serial="348236533430"
Found DFU: [0483:df11] ver=2200, devnum=30, cfg=1, intf=0, path="1-4", alt=1, name="@Option Bytes  /0x1FFFC000/01*016 e", serial="348236533430"
Found DFU: [0483:df11] ver=2200, devnum=30, cfg=1, intf=0, path="1-4", alt=0, name="@Internal Flash  /0x08000000/04*016Kg,01*064Kg,03*128Kg", serial="348236533430"

Flash the black pill with the F411 chip
$ sudo dfu-util -d 0483:df11 -a 0 -s 0x08000000:leave -D firmware_F411ce_C2.bin

------------------------------------------------------------------------------------------------
 
Flash the ESP32 with the esptool (Python script) and an FTDI (USB2UART) converter, not connected to the Vdmot board

esptool --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 ./VdMot_Controller-master/software_esp32/bootloader_dio_40m.bin 0x8000 ./VdMot_Controller-master/software_esp32/ESP32_partitions.bin 0xe000 ./VdMot_Controller-master/software_esp32/boot_app0.bin 0x10000 ./VdMot_Controller-master/software_esp32/ESP32_firmware.bin

closed 2023-01-30 by bluephoenix28