Lenti84/VdMot_Controller #62: isr calls callback right when isr_target reached but motor might still spin

QuadCorei8085, 2023-02-14:
I don't have your board or debugged the code just went through the schematic and the codebase.

Assuming I issue a few steps of open/close to the motor 1000-100000x would'nt this approach lose the correct position after a while?

I'm assuming the rotor counter works to a certain speed (amplitude gets lower) so at lower speed this loses the steps which I believe negligable (?) effect if motors are operated in ON/OFF mannger.

But also the ISR quits counting right when reaching the target, any DC motor in my experience will still rotate in freewheeling mode (don't know how much here we are talking about but the error will be accumulated). Now i'm certain it wont be simmetrical in both ways so this counting up/down will eventually cause a drift error in the step following.
Would'nt it be more precise to wait till the ISR/tacho impulses "ramp down"? Once target is reached the counting continues for 100-500ms you would have more accurate position tracking.

If I saw it correctly the opening and closing states detect endpositions thus sooner or later this would be noticed.

Lenti84, 2023-02-16:
If you don't have one, get one ! ;-) Just kidding...

You are right, moving the motor very often will give worse accuracy.
This fact is handled by two things:
1. You can configure the controller to do a calibration cycle depending one the weekday. Max. is 1 cycle each day.
2. You can define the count of movements of the motor, until a new calibration cycle is performed automatically.
This helps to keep accuracy up.
[image]

Additionally the normal usecase is to use the controller for valves, which are controlled by an overlying temperature controller. This will compensate small deviations between target and real position.

Motor is operated only on/off. No linear regulation. So BackEMF is most of the time at a high level and triggering the counter.

I never measured the time or the signals of the motor in the moment it is turned off. But I would also think there is some movement even the controller has shut off the current. So behaviour should be asymmetric, yes.
And it is additionally asymmetric, because in closing direction the spring in the valve helps closing, but in opening direction the motor has to push against the spring.

I will put the idea with counting until the signals are lost after motor turn off on my list. Maybe this will be part of the software in further versions. Thanks for your suggestion.

Info: for the typical VdMot valve the counter counts around 7500 pulses for a full movement of the motor. This value is measured each calibration cycle and then used to calculate from % movement to counts.

QuadCorei8085, 2023-02-16:
Oh nice wasn't aware of this featuer, well thought through.
I already have a thermostat running on esp32 just checking the motvd option as it is tempting to have a settable min flow rate to keep the floor semi warm and have a faster reaction time for the heating.
I think i will use an ina219 and just rely on current measurement, i don't need exact position as the min flow will regulate on pipe temp and for heating i think the valves will be opened fully, or maybe  regulated by +/- steps based simply on time.
The rotor counting seems too complicated (compared to the simple current based start/end position finding + time based steps) and somehow I feel (regardless how awesome it looks) it looses the point in my case where the pump can not pressurize all 7 loops and based on how many of them are open the flowrate will fluctuate.

closed 2023-02-16 by QuadCorei8085