Lenti84/VdMot_Controller #63: question - vdmot currennt graph

QuadCorei8085, 2023-02-18:
I'm only opening an issue because discussions are not enabled.
I ordered 2 vdmot to play around and tried one of them on my distributor.
I'm not familiar with various distributor or valve types, mine is open (full flow) when the needle is not pressed in (default). The wax valves are pushing them in in cold phase and when heated up the wax expands and releases the pressure.

So my issue is that in my case I can not really identify anything from current measurements. (vdmot top=fully retracted)
Here is a valve closed -> opening sequence:
[image]

The case of the valve is fully opened (vdmot starts from top position) -> close:
[image]

If I see it correctly in you case the vdmot push pin has a play/gap so you have a freewheeling ~18mA phase, then a contact period, and a definitive section where the pin is pushed all the way in.
I guess you also can't tell when the flow starts/stops just from end positions of the motor?
I also found that there is a huge "hysteresis" between opening and closing when microstepping the motor with 20-50ms impulses.

Also funny how much smoother your current measurement is, I used the PPK2 from nordic. You can literally see the commutation:
[image]

Lenti84, 2023-02-18:
Did you had a look at https://github.com/Lenti84/VdMot_Controller/blob/master/system/actuators.md ?
There you will find my measurements from the beginning of the project.

The water flow normally can not be determined by the current consumption of the valve motor.
Maybe only if you have enormous pump pressure and the valve needle must be pushed against that force. But i dont think this is realistic.

Actual software only can detect endpoints of the valve needle by drastic increase in current consumption.
Additionally the commutations are counted to derive a position from 0...100% between the two endpoints.

Some idea i have in mind is to detect the moment, where the valve motor (vdmot) hits or leaves the valve needle (around 2600) and the point, where the needle compresses the rubber seal inside the valve (around 6300).
Both points can be observed in the current curves from https://github.com/Lenti84/VdMot_Controller/blob/master/system/actuators.md.
Both points would give a new 0...100% range for valve movement, which is nearer at the truly arising flow.
[image]

QuadCorei8085, 2023-02-19:
Yes I saw your measurements that is why I asked the question.
I believe mine looks like the motor always have a contact with the needle hence there is no ~15mA region in my measurements. 
Also the rise at both ends looks different in my case especially when the vdmot hits its own end position.
In my case the current rises quite quickly, and rises much slower when compressing the valve.

What I meant is the actual valve authority (0-100% flow) is a much narrower band than the motor current given end-points. And there is a huge deadzone in my case on both ends. Finding the end position (vdmot fully extended - valve fully closed) is relatively easy but how much I need to step back is not trivial, as I will not know when the water starts to flow or stops.

I wonder how the homematic and other systems handle this.
It simply sounds illogical to me just to measure the endpoints of the motor.
Somehow that also needs to be calibrated or measured via other methods what is the flow.
Or is the flow in your case 0-100% in the calibrated range?

SurfGargano, 2023-02-19:
Pls see also https://github.com/Lenti84/VdMot_Controller/issues/61

0-100% is calculated in the calibrated range.

closed 2023-02-23 by QuadCorei8085