Lenti84/VdMot_Controller #75: VdMot WT32 crashes repeatedly

elektronikworkshop, 2023-02-27:
Here's the backtrace. Same behavior with 1.0.7 and 1.0.8 - here's the backtrace

```
[...]
pop gvlon 255 
pop gvlst 
Reconnecting MQTT...

Stack smashing protect failure!

abort() was called at PC 0x400ef5d4 on core 1

Backtrace: 0x40082fe1:0x3ffb1fc0 0x4008d33d:0x3ffb1fe0 0x40091f8d:0x3ffb2000 0x400ef5d4:0x3ffb2080 0x400d91b5:0x3ffb20a0 0x400d950f:0x3ffb2150 0x4015855b:0x3ffb2170 0x400de261:0x3ffb2190 0x400d9982:0x3ffb21d0 0x400d7eb6:0x3ffb21f0 0x400d8336:0x3ffb2210 0x400deb9b:0x3ffb2230 0x400de8fa:0x3ffb2250 0x400d8d26:0x3ffb2270 0x400e8d61:0x3ffb2290

ELF file SHA256: 0000000000000000

Rebooting...
VdMot_Controller
Start Config
SPIFFS booted
get file : STM32_C2_firmware.bin
Interface type 0
Setup Eth 0
Server started 0

ETH Started
Setup Wifi 0
Interface type 0
Setup Eth 1
Server started 0
Setup Wifi 1
[...]
```

It runs for some seconds and crashes again most probably on "Reconnecting MQTT". This device runs on WiFi.

I had the same problem yesterday evening. It may be an coincidence or not but it seems only to happen in the evening/night on this device - so perhaps it's related to some date/time code. If I'd have to name a time: I have discovered it around 22:00. But take this last paragraph with a big grain of salt.

elektronikworkshop, 2023-02-28:
Just had another device doing this. It was even worse - it over drove my valves until they clicked (I'm an engineer myself and I don't like it if stuff gets tortured - arrrgh), so it seems that perhaps the STM32 crashes or is in a bad state as well. It exhibited the same symptoms on the status page as well. This device runs on WiFi too, but I'm not sure if that's a contributor... I'm just trying to collect data

SteffenB71, 2023-02-28:
Hello
Can also confirm that the 1.08 also leads to this strange behavior of the actuators for me. Back to 1.07 and everything works again.
Steffen

elektronikworkshop, 2023-02-28:
I had those crashes with 1.0.7 and  1.0.8.
I just discovered that one crash was so bad that it even popped one of the motors of its valve. So the STM32 must have been in a desolate state that it was deadlocked and not monitoring any current.

I will keep observing. I have three on ethernet and two on wifi. So a good test farm.

Lenti84, 2023-02-28:
Thats a weird behaviour... Hmmm

I remember problems with WiFi an MQTT in older versions. ESP32 crashed while trying to connect to the broker. Reason was around wifi setup command WiFi.mode(WIFI_MODE_STA); .
Can you narrow down, if it happens only on ETH or Wifi or on both equally?

I have my two controllers connected to mosquitto via ETH. Running fine with mostly latest software since 25d...

Status page looks very strange. But I have little idea why the STM32 drives the valve into end stops without stopping.
Normally STM32 controls valves independently from ESP32. But maybe it gets corrupted too.

There is a command, which sets the threshold of the current comparator for end stop detection and tehrefore motor stop. If this command is send unintentionally with wrong values STM32 will forcefully drive motor beyond end stops.

Or maybe STM32 gets busy by communication overhead on UART interface and cant stop due to overload. No idea if this can happen in reality. But valve motor control loop runs at same speed of communication loop.

Can you please try to use an older version of the software? 1.0.5 for example?

Software 1.0.8 has a changed reset mechanism, which leed to problems during development, but was fixed at the end. At least it seemed so. Flag -DforceHardReset in ESP32 source will switch behavior back to old STM32 reset handling.

There is also a debug interface. If you setup an SysLog server, you can log several messages of the ESP32 during runtime. I'm using visual syslog server 1.6.3 for example. Maybe this helps to collect more data.

elektronikworkshop, 2023-02-28:
Hi Lenti,

thanks for commenting on this issue. Such stuff happening in the field is like Murphy's birthday party.

Yeah, I'm currently waiting for more data. The two devices on which this happened are on WiFi. I had another device on ethernet with borderline behavior but I'm pretty confident that this was due to driving it with a to weak of a power supply and causing brown outs :) That's not possible with the two problematic ones which are powered by beefy upcycled router power supplies.

Yes, for some apocalyptic motor overdrive the STM has to be dead. No idea what caused it (some ideas though of course - you are pointing in the same direction). This shows that some kind of watchdog and/or high priority code would be a good idea. A mechanism which guarantees that the current monitoring never stops (ADC or timer interrupt with the interrupt priorities set correctly, so that it can't be choked by the UART for instance - in contrast to what would be the case in a traditional main loop) or at least cuts the power to the drivers in case some abnormal condition (e.g. when the watchdog barked) has been detected.

I will keep the 1.0.8 running to check if I can see a pattern. If it occurs again I will switch to ethernet and check if I can reproduce it on ethernet as well.

I have several projects running on WiFi connecting to MQTT brokers and providing telnet interfaces and more - on both ESP8266 and ESP32 and I never had problems with producing stack corruptions on MQTT (re-)connects. If I can reproduce a pattern which points to the networking code, I will audit it - if I'm not too busy. Same for syslogging...

Regards
EW

P.S. I have some other (hopefully good but for sure always constructive) ideas for VdMot, but I will address them in dedicated tickets.

Lenti84, 2023-02-28:
To help debugging I have extended the code a little.
- ESP32: added Uptime counter, this value is published via mqtt topic /common/uptime, maybe this helps to see when system crashes
- STM32: max read length for UART buffer is limited to 990 now, not unlimited anymore, can help against erroneous data floods
- STM32: valve_loop() is now called via timer interrupt, this hopefully ensures correct motor drive and stopping during high system loads

Please see: https://github.com/Lenti84/VdMot_Controller/commit/05eb9992e0b95821455918e9bf9bfc8b0c74547e
Also uploaded some binaries.

I will have a look at the other tickets a little later. But meanwhile thanks for your suggestions.

elektronikworkshop, 2023-02-28:
That's great, thanks!

And there can't be enough guard rails to keep the STM safe. I thought about it just some minutes ago: A huge load of commands can easily be produced by an ESP32 going bonkers by booting, crashing, booting, crashing, ... within seconds. During each boot the STM gets a bag load of queries which then quickly would overwhelm the MCU. I'm not familiar with the code base - so I don't know for instance if the ESP resets the STM on each boot or if the STM continues to run. Latter would render the theory more plausible though.

Don't worry about the other tickets - it's just to get them off my ToDo-lists and others can think about them and perhaps develop even better ideas.

elektronikworkshop, 2023-02-28:
Yet another thought: given the STM continues to run: this would confront the ESP after a crash-reboot with all the traffic generated by the STM - which could then end up in a vicious cycle. But again: this is based on a free running STM. Synchronous boot and reset is for sure preferable as it makes a complex system much more predictable and cuts millions of corner cases...

elektronikworkshop, 2023-03-01:
Tonight around 1:00am adds one more data point to night time crash list. The ESP didn't seem to have entered the crash-reboot-loop this time - at least I wasn't able to see it on the serial when I arrived at the crash site. The serial was completely silent but it was set up correctly despite the hurry because it later logged the brown out when I finally pulled the plug.

But before pulling it I tried something different: I reset the STM manually via the NRST button. This stopped the madness and the system seemed to have recovered. I unplugged it anyways for the sake of better sleep without worrying too much about my actuators.

[image]

And I have two more interesting samples:

#### 1)
This morning I found that two other devices crashed.

The interesting thing is: they crashed at around the same time (8:00 am). This could point to problems with some date/time/ntp code. There's nothing special happening at 8:00 am from my side AFAIK. The switch of the two wifi devices from "on" to "manual" happens at 8:00 pm and the other way round at 9:00 am (some minutes ago).

Both are on 1.0.8. _but one is on ethernet_ (!), and the other on wifi. So the data situation for the pure wifi theory gets thinner.

Both seemed to have recovered without entering the loop of death.

To keep stuff documented: On the ethernet devices I actively disabled the wifi in the configuration. The wifi devices are on "auto" (in case something goes horribly wrong and I have to bootstrap them via ethernet).

#### 2)
And a controller on ethernet with the combination of ESP@1.0.7 and STM@1.0.5_C2 seems to run since almost four days now. The strange firmware combination seems to stem from my flashing experiments a week ago when I had a 1.0.5_C2 lying around during the assembly for testing my DFU flashing setup. I forgot about it but discovered it this morning.

elektronikworkshop, 2023-03-01:
Oh, it's so annoying. Exactly at 22:00 one device crashed again. It's on 1.0.8, here's the output when it's in the loop of death - nothing new though.

```
pop gvers 
pop gmotc 
application setup start position finished
pop gtlnm 
pop gonec 255 
pop gvlon 255 
pop gvlst 
Reconnecting MQTT...

Stack smashing protect failure!

abort() was called at PC 0x400ef5d4 on core 1

Backtrace: 0x40082fe1:0x3ffb1fc0 0x4008d33d:0x3ffb1fe0 0x40091f8d:0x3ffb2000 0x400ef5d4:0x3ffb2080 0x400d91b5:0x3ffb20a0 0x400d950f:0x3ffb2150 0x4015855b:0x3ffb2170 0x400de261:0x3ffb2190 0x400d9982:0x3ffb21d0 0x400d7eb6:0x3ffb21f0 0x400d8336:0x3ffb2210 0x400deb9b:0x3ffb2230 0x400de8fa:0x3ffb2250 0x400d8d26:0x3ffb2270 0x400e8d61:0x3ffb2290

ELF file SHA256: 0000000000000000

Rebooting...
VdMot_Controller
Start Config
SPIFFS booted
get file : STM32_C2_firmware.bin
Interface type 0
Setup Eth 0
Server started 0

ETH Started
Setup Wifi 0
Interface type 0
Setup Eth 1
Server started 0
Setup Wifi 1
192.168.1.63
application setup finished
Start services
push gvers 
push gmotc 
push gtlnm 
push gonec 255 
push gvlon 255 
push gvlst 
pop gvers 
pop gmotc 
application setup start position finished
pop gtlnm 
pop gonec 255 
pop gvlon 255 
pop gvlst 
Reconnecting MQTT...

Stack smashing protect failure!

abort() was called at PC 0x400ef5d4 on core 1

Backtrace: 0x40082fe1:0x3ffb1fc0 0x4008d33d:0x3ffb1fe0 0x40091f8d:0x3ffb2000 0x400ef5d4:0x3ffb2080 0x400d91b5:0x3ffb20a0 0x400d950f:0x3ffb2150 0x4015855b:0x3ffb2170 0x400de261:0x3ffb2190 0x400d9982:0x3ffb21d0 0x400d7eb6:0x3ffb21f0 0x400d8336:0x3ffb2210 0x400deb9b:0x3ffb2230 0x400de8fa:0x3ffb2250 0x400d8d26:0x3ffb2270 0x400e8d61:0x3ffb2290

ELF file SHA256: 0000000000000000

Rebooting...
VdMot_Controller
Start Config
SPIFFS booted
get file : STM32_C2_firmware.bin
Interface type 0
Setup Eth 0
Server started 0

ETH Started
Setup Wifi 0
Interface type 0
Setup Eth 1
Server started 0
Setup Wifi 1
192.168.1.63
application setup finished
Start services
push gvers 
push gmotc 
push gtlnm 
push gonec 255 
push gvlon 255 
push gvlst 
Reconnecting MQTT...

Brownout detector was triggered

VdMot_Controller
Start Config
SPIFFS booted
get file : STM32_C2_firmware.bin
Interface type 0
Setup Eth 0
Server started 0

ETH Started
Setup Wifi 0
Interface type 0
Setup Eth 1
Server started 0
Setup Wifi 1
192.168.1.63
application setup finished
Start services
push gvers 
push gmotc 
push gtlnm 
push gonec 255 
push gvlon 255 
push gvlst 

abort() was called at PC 0x400ef6a5 on core 0

Backtrace: 0x40082fe1:0x3ffbe9cc |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
VdMot_Controller
Start Config
SPIFFS booted
get file : STM32_C2_firmware.bin
Interface type 0
Setup Eth 0
Server started 0
```
I'm not 100% sure if the brown out log corresponds to my un- and re-plugging the power because it was so annoying how VdMot tortured my actuators. Excuse me - I really had a bad day and this just gave me the rest.

But the timing of 8:00 and 22:00 can point to some MQTT subscription stuff. Because at 22:00 and 8:00 I switch to and from night time temperature reduction. I do that by adding the reduction to the temperature sensor data. I just logged it and this is what gets TXed:
```
indoor/ba2/og/plan-sura/temperature : msg.payload : number
19.810000000000002
```
The funny rounding is some artifact from Node-Red - I don't change it now because any MQTT client must be able to handle any message the broker delivers - even invalid data.

Perhaps this payload overflows a buffer in the subscription callback routine of `<blabla>/tValue`? Let's check...

### There we go:

My bet is on https://github.com/Lenti84/VdMot_Controller/blob/dcdbeada24d6e19d4e978b64fdf78c6a4e9407f2/software_esp32/src/mqtt.cpp#L257-L258

You don't protect memcpy's target
https://github.com/Lenti84/VdMot_Controller/blob/dcdbeada24d6e19d4e978b64fdf78c6a4e9407f2/software_esp32/src/mqtt.cpp#L198
against arbitrarily long payloads. I refrain from commenting ... let's stay constructive :)

This is the poster boy of software security (and stability). It's perfect for corrupting not only the stack - It would be a piece of cake to inject and run code on the VdMot by using the tValue MQTT payload.

Ok, how to fix it? Apart from being so stingy with the `value` buffer size (I would at least give it 32, it's on the stack anyways so who cares?), I suggest something like the following (illustrative):

```c++
memcpy_s(value,
         sizeof(value) - 1,  // keep zero termination intact
         payload,
         length);
```
Or if you'd like to stick with `memcpy`:
```c++
memcpy(value,
       payload,
       std::min(sizeof(value) - 1, length));
```

Ok, get this comment out - I will edit if there's something to amend.

@SurfGargano owes something to me. At least 10 liters of beer:
https://github.com/Lenti84/VdMot_Controller/blame/544de769b9e40b6fc1b11078ff790ce9bc06b8cc/software_esp32/src/mqtt.cpp

And on top of that he has a homework assignment: someone should review the whole code base for buffer overrun flaws like this one. Search for use of similar style functions like `strcpy`: zero termination is never guaranteed, i.e. always use the ' safer' counterparts like `strncpy`  etc.. And educate yourself about safe coding... Ok, that's enough pedagogy - but after finding such an annoying bug I have a lot of pedagogy credits :)

elektronikworkshop, 2023-03-01:
I will provide a pull request tomorrow - I'd like to cash in the credits of this bug fix :)

This excludes the homework assignment of course.

elektronikworkshop, 2023-03-01:
The improvements and safeguards on the STM remain valid though. This bug perfectly highlighted the previously discussed shortcomings, which you @Lenti84 already addressed partly. I hope I don't have to remind you of race conditions when exchanging data between main loop and interrupt routines. Better safe than sorry...

elektronikworkshop, 2023-03-01:
https://github.com/Lenti84/VdMot_Controller/blob/dcdbeada24d6e19d4e978b64fdf78c6a4e9407f2/software_esp32/src/mqtt.cpp#L212-L213

therefore

```c++
uint16_t beer_in_l_to_ew = 10;
beer_to_ew *= 2;
```

Lenti84, 2023-03-02:
Great finding! 
Thanks for your contribution in finding and fixing.

I merged your pull request, but to the developer branch.

You can find the updated ESP binary under https://github.com/Lenti84/VdMot_Controller/blob/developer/releases/1.0.9_Dev/ESP32_firmware.bin
It is running at my system at the moment. Also MQTT reception works. At least for valid data.

Further shortcomings will be adressed in the next time.

elektronikworkshop, 2023-03-02:
You're welcome!

I just installed the 1.0.9_dev release and deployed some torture message generation:
```javascript
var ntr = flow.get("night_time_reduction");
if (ntr > 0) {
    var t = parseFloat(msg.payload) + ntr;
    msg.payload = t + 0.00000000007;
}
return msg;
```
aka
```
20.630000000069998
```
And the system is still running ... :relieved: ... It runs the STM 1.0.9_Dev_C2 as well.

I will close this issue as I'm pretty confident that the root cause of this particular symptom is fixed (i.e. the stack corruption). It's not said that there are other unbound copies or alike coming to haunt us in the future. Perhaps it makes sense to open a dedicated "homework" issue to keep this code review on the list. It could safe us from a lot of potential headaches ...

commit aed73087 (2023-03-02): * Fixed buffer overflow by unbound memcpy (closes #75)
* Removed redundant code
* Reduced scope of `value`
closed 2023-03-02 by elektronikworkshop