Lenti84/VdMot_Controller #77: Adjustment indexing with linked valves probably wrong

elektronikworkshop, 2023-03-01:
I just discovered a strange behaviour:

I wanted to lower the current opening percentage of tht valves. control is on, but the temperature is almost there but the valves were still at 50% because this device crashed (I learned though that manual entry doesn't affect the control loop though).

Valve 1 is the main valve, all others are linked (slaves).

When I set now set all the new values to 10, every valve except for 2 (the first slave) got set to the new value. This got me suspicious: When I now change only the first slave it affects the second slave.

There is no way to set the first slave. It looks like the indices are messed up when linked valves are involved.

I tried to capture it in two comprehensible screen shots:
[image]
[image]

SurfGargano, 2023-03-03:
I assume the Heat Control in the control page was not set to Manual ? 
This is not the correct way to do this. If you want to set the valves manual, you have to set the Heat Control to manual first.

elektronikworkshop, 2023-03-03:
Hi!

I can not recall it, but usually I have them on manual and vdmot gets its state via MQTT.

If you as UI designer think that this is not allowed, than you should disable this UI option when vdmot is not in manual mode.

Regards
EW

closed 2023-04-28 by SurfGargano