Lenti84/VdMot_Controller #91: MQTT als JSON ?

TTr0n, 2023-07-31:
Die MQTT Werte, die an den Broker gesendet werden, kommen als reine String Werte an.
Wäre es möglich, eine Option hinzuzufügen, die die Daten als JSON Paket schickt?

LG

SurfGargano, 2023-08-05:
Nein, es sind ja meistens Einzelwerte. Da bringt JSON nichts.
Es ist aber möglich über das JSON-Interface Daten zu schicken oder abzufragen.

TTr0n, 2023-08-05:
Abfragen/Pollen wollte ich vermeiden. Ein Subscript auf das Interface ist wahrscheinlich nicht möglich, oder doch?
Reine Strings sind aber auch nicht so toll bei den MQTT Daten. Ich weiß nicht wie das im IO Broker funktioniert, aber die Einzelwerte als JSON formatiert (und nicht als String) sind mMn auf dem MQTT Server viel besser zu verarbeiten.

SurfGargano, 2023-08-05:
Hatte ich auch nicht vorgehabt. In iobroker , fhem und andere mqtt Broker wird dies als Einzelwerte verarbeitet und dabei bleibt es auch.

closed 2023-08-05 by SurfGargano