Lenti84/VdMot_Controller #96: Wrong Port speed in Manual/Pictures for flash ESP32

Alfons-Hetler, 2023-09-22:
For flash the ESP32 Board is a wrong speed in the online Manual.
That was the reason why i cant initial flash the Board.

Lenti84, 2023-09-24:
I think there is no right or wrong baudrate for the "flash download tool" as shown in https://github.com/Lenti84/VdMot_Controller/blob/master/assembly_kit/installation.md

I have already used successfully a lot of different baudrates, like:  230400, 460800, 576000 or 921600.

But my experience is also, that sometimes it will not work and after changing the baudrate it does.

closed 2023-09-24 by Lenti84

Alfons-Hetler, 2023-09-24:
I try it too and it did not worked.
It says: Custom element doesn´t exist: plothy-graph.
I deleted the app and begin from new but the error stay there.