Lenti84/VdMot_Controller #98: No issue, create a Telegram group for Q and A like Tasmota or AVM

Alfons-Hetler, 2023-09-22:
If we have questions we can help us together. And we can maybe support the Project for more Members.

It will helps all.

Lenti84, 2023-09-24:
Do you had a look at
https://github.com/Lenti84/VdMot_Controller/discussions
or
https://github.com/Lenti84/VdMot_Controller/wiki
?

Alfons-Hetler, 2023-09-24:
Yes but thats no Telegram Group ;-)
A Telegram Group can help people really fast and we can support the Developer very quick and easy.

Alfons-Hetler, 2023-10-20:
I know all the Websites already. Thanks for sharing information. But if we have a chat or a channel on smartphone i think we can help faster and simply. We can share Pictures etc. it is only a better support Medium i think. It is only an extension to the existing possibilities.

Lenti84, 2023-10-25:
Dear Alfons,

I just created a Telegram Group for VdMot Controller. Feel free to join and try.

Everybody else is welcome too.

https://t.me/+P0_VMEDQ8NMzZTUy

Kind regards

Alfons-Hetler, 2023-11-12:
Wollen wir das noch offen lassen zur Info? Ansonsten ist das ja quasi erledigt und klappt gut ;-)