Lenti84/VdMot_Controller #104: Some questions, power usage, wires, KNX

shivan, 2023-10-19:
Since I did not find another way for contact.

1) What's the power consumption in "standby" so when no motor is moved?

2) Do you use all 4 wires of the VDMOT? I read something that two wires are for the motor and the other two can be used for better positioning?

3) Did you also think about a possibility to connect this to KNX? (currently I do not have LAN there, but KNX 😄  )

SurfGargano, 2023-10-21:
Power consumption is about 1.2 W in standby
Only 2 wires are used for the motor, since the motor has only 2 wires connected. I have extended 2 motors with a shielded KNX-cable by 10 m.
You can use Wifi , but the first (Wifi) installation has to be done by cable.

KNX : Andreas will answer.

Lenti84, 2023-10-25:
I am thinking of a KNX Support for a longer time...

I just created a poll / discussion to find out how interessted the community is:
https://github.com/Lenti84/VdMot_Controller/discussions/108