Lenti84/VdMot_Controller #113: [1.3.0_Dev] MQTT tTarget only read after boot?

ei-ke, 2023-11-12:
Moin zusammen,

I recently upgraded from 1.0.7_Beta via 1.2.2_Dev to 1.3.0_Dev.
Not sure when on the way it happened or if I'm the only one affected, but changing tTarget via MQTT after the valve calibration run that happens after booting the device has finished (maybe even earlier) seems not do be noticed by the device.

I can clearly see the changed tTarget (coming from Home Assistant) on my broker but the controller just ignores it. If I'm rebooting if from the webinterface the new tTarget is picked up correctly.

From 18:31:00 to 18:34:00 tTarget for "Bad" should have been 19.0 and not 18.5.

Kindly let me know if you need further informations.

PS: I also tried different `Interval (ms)` and `Publish Interval (s)` values without any success.
Eike

<details>
<summary>
Screenshots and atomic syslog
</summary>

[image]
[image]
[syslog.txt](https://github.com/Lenti84/VdMot_Controller/files/13328740/syslog.txt)

</details>

SurfGargano, 2023-11-12:
The interval (10000  ms) is too long. Recommended and also default is 1000. This interval is the time the controller looks for new incoming data and for possible sending data. You can activate publish on changing to reduce traffic.  (In this case you can set publish interval to 60) For home assistance. 
 you should also set separate topics. Pls see the Wiki /mqtt for documentation

ei-ke, 2023-11-12:
Thank you for the hint. Setting the interval to 1000ms fixed the issue. Not sure when I made this mistake.
I'll have a look at the "separate topics" settings tomorrow.

closed 2023-11-12 by ei-ke

Alfons-Hetler, 2023-11-12:
Komm mal im Telegram Chat dann können wir mal schauen was das Problem ist. Bei mir gehts ohne probleme.