Lenti84/VdMot_Controller #123: minor observations with 1.3.6 version

harvey637, 2024-03-23:


harvey637, 2024-03-24:
some more feedback:
status icon: with error ther is an red error mark, thats good!
without error: no icon, just the square, where the icon should be :-(

Calibration: when will an automatic calibration start after reboot?
Manual calibration: status show "calibration" for some time, but nothing happens. After some time it goes back to "connected",
no open/close/idle or movment to calculated position (in the next 30 minutes or so ....

cu
Harvey

harvey637, 2024-03-30:
thank you for the calibration info!

some more ideas: after some findings in my graphs (no response to api call) I found out the mqqt status: WDT, which sounds like WatchdogTimer. Sadly no local log an no Info on the reason ...
In the GUI there is not hint about this.
Maybe it is an idea to put the last status in the detailed "status" page, on the right, below calibration?

Some thought on documentation: I guess, the "hour time" in calibration means to recalibrate after xxxx hours? hours=0 to ignore? Is hours a good scale to calibration, better as "days"? ... just thinking ....
cu
Harvey

SurfGargano, 2024-03-30:
hour time means time of the day. If you enter 0 , the calibration starts at midnight of the activated day.

If you set the interval to 60000 ms, the VDMot Controller checks every 60 sec if there are new values on mqtt. This will result in a long time response.
With keep_alive 45 seconds the connection will be closed aftrer 45 seconds after the last transmit, thats why after 2 minutes a reconnect occurs.

Set the  interval to a smaller value ( I have 1000ms, results in a faster repsonse)  and the keep_alive to a greater one. ( I have set to 120 seconds)

best regards Werner

harvey637, 2024-03-31:
Hi SurfGargano
thank you for the answers, I am happy with the interval/keep_alive settings now, ok!

Calibration timing on the homematicIP happend immedeately after reboot, so it was better predictable. (power off, power on, watch for the valves to move for calibration).
With vdmot if I am in front of the valves I have not idea, when the first move will initiate the calibration. But this is my minor problem, may be I have to solve it with browser on vdmit on my mobile :-)))

The info about "hour_time" is ok, but in contrast to the wiki. it says:
h - After choosen hours, all valves are newly calibrated by the controller.
Days - Choose the days, when all valves should be newly calibrated by the controller. Calibration happens at midnight.

Maybe a refresh on that info clarifies settings :-) I guess that hour_time is calibration time for every selected day ....

by the way.....
- how to reset counters (without full reset ...)?
- think about showing the "cr" (calibration retrys?), the "link" and the "controlactive" flags for each valve in extended status?

Now this is only in the json api-call...

happy easter!
cu
Harvey

closed 2024-08-06 by SurfGargano