Lenti84/VdMot_Controller #130: STM32 cannot be flashed !

valig80, 2024-11-01:
Hello,

I am creating this post for asking help regarding the flashing of the STM32 Black Pile.

Today I soldered all the components on a VDMOT board (Revision C3) and as peer software installation instructions I managed to flash the WT32. Then from the GUI I tried to flash the STM32 with latest firmare version 1.4.2 (C2) but I always get an error...

For flashing the STM32 I have the X20 jumper set, and I am keeping the BOOT switch pressed. I get initially the message "STM32 is updating and while keeping BOOT pressed the progress goes till 10% and thereafter it shows Error
[image]

Since this is my second controlled I tried also following, I've inserted the STM32 into my first controller which is working perfectly, and I tried to flash the STM32 through its GUI.  Also, there the flashing process is the same, it ends with Error.

Are there any possibilities to push the firmware intro the STM32 (through the usb-c or via USB-Serial to PINs) ?
Or since the Chip ID is shown always with 0x0 can it be that the STM32 is faulty (the LEDs, switches are working normally)  ?

Thank you in advance for a fast answer.
Val

SurfGargano, 2024-11-01:
Do you get the Stm32 from Andreas or buy it by your own?

valig80, 2024-11-01:
> Do you got the Stm32 from Andreas or buy it by your own?

I bought the entire assembly kit from Andreas

SurfGargano, 2024-11-01:
It seems that the stm is defective. There are some not functional clones on the market. 
https://hackaday.com/2020/10/22/stm32-clones-the-good-the-bad-and-the-ugly/

SurfGargano, 2024-11-01:
Btw, you can try 1.4.3 (wth) . With boot0 flashing the stm  it reduces the baudrate, if an error occurs.

Lenti84, 2024-11-02:
Hi together,
latest orders were shipped with preprogrammed stm32 boards. Programming of board was done via STLink utility. Programmed version ist 1.4.4 as written on the antistatic plastic bag.

Does the LED of your board flashes?
Does it run in standanlone mode via USB cable?

Kind regards 
Andreas

valig80, 2024-11-02:
Hi Andreas

Sorry, I did not read what was on the plastic bag and therefore I didn't realize that the STM32 is preprogrammed.

When I connect the STM32 through the USB-C on my connector, I get the red LED (Power) on all the time and the blue LED (C13) flashing fast initially and after 7 seconds it blinks slowly at every 3 seconds.

If the STM32 is plugged in the VdMot controller, I get both LEDs blinking as described above and on the GUI I see following:

[image]
[image]

Today I will get another STM32 chip and an ST-LINK programmer. I plan on doing the following:
1) insert the new STM32 chip in the VdMot controller and try to flash it from GUI
2) Try to flash the old STM32 chip somehow with the ST-Link, but here I don't have the knowledge yet as I only played before with ESP32s

best regards,
Valeriu

valig80, 2024-11-03:
Hello

I am posting some updates. Today I received by Amazon another STM32 and an ST-Link programmer.

Unfortunately, I did not have any luck on making the second VdMot controller to recognize  STM-32 chip.
I tried following:

1) Check if the VdMot controler nr2 (#2) recognizes the STM32 chip from the VdMot nr1 (#1)
Since I have two VdMot controllers, I swapped the STM32 from VdMot #1 into VdMot #2 and the VdMot #2 reconizes the STM32 (#1). Therefore at this point, I assume that the new soldered VdMot #2 controller is not faulty

2) Tried to flash the STM32#2 into VdMot#1, it fails also exactly like when flashing with VdMot#2

3) Inserted the new received STM32 (#3) from Amazon into VdMot#2 and tried to flash it.
The flashing in Boot partition works and the GUI shows the Chip ID but after verification and 100% completion the STM32 remains only with the power LED on and it is not recognized in the GUI. Flashing it again in Normal partition fails

4) Flash of STM32 #2 and STM32 #3 with ST-Link utility
With ST-Link I am able to read both STM32 chips and I am also able to program them. However, when I am inserting the STM32 chips in the VdMot #2 both chips are not recognized in the GUI

5) I flashed different version of software also on WT32 and also on STM32
I tried with version 1.3.7 and also with 1.4.3. BTW Andeas you mentioned that the STM32 was preprogrammed with 1.4.4, where can I find this version ?

At this point I am out of ideas, what else can I try?

valig80, 2024-11-08:
Hello

I managed to solve the issue with the STM32 and now the VdMot controller is fully functional. :)

During the several programming attempts through the controller itself and also through the ST-Link I observed that the unrecognized STM chips were of this type: STM32F401xB/C.
The STM32 chip from the functional controller was an STM32F401xD/E, and therefore I ordered again another STM32 chip from Amazon but this time from another supplier (AZ Delivery)

It arrived, I flashed the firmware_c2 through the ST-Link utility and after plugging the STM in the controller board everything worked perfectly.

Therefore, I assume that only the STM32F401xD/E chips are at the moment usable with the controller.

closed 2024-11-08 by valig80

ralf-e, 2024-11-20:
Hello,

could this be a memory problem?
What is the minimum memory size required for the STM32F01 by VdMot?

STM32F401CEx: 512k Flash and 96k RAM,
xx401CDx: 384k/96k
xx401CCx: 256k/64k
xx401CBx: 128k/64k

SurfGargano, 2024-11-21:
No, the required ram/rom is 
RAM:   [=         ]   7.5% (used 4908 bytes from 65536 bytes)
Flash: [==        ]  20.5% (used 53776 bytes from 262144 bytes)