Lenti84/VdMot_Controller #132: Two ports not working

padarom, 2024-11-11:
I recently ordered and assembled two kits which I intended to use with 11 valves each. The first kit went smoothly, the second one showed me some strange behavior when trying to install and set up all the valves.

Took me a while to isolate the issue, but it turns out that:
- When connecting any valve to X3, the controller thinks all other valves are blocked. The web interface slowly changes all their states to "blocked", even though a second ago they were still functioning
- Same behavior for X4
- Also for X4: When scanning for valves, it shows X2 as connected, even if X4 is the only valve connected to the controller. When only connecting a valve to X2 it operates correctly.

For now I am running it with X3 and X4 disconnected, but I'm wondering about what caused this. Is it somehow obvious if this is a PCB/BluePill/... issue, or does this not make sense for anyone else either? I rechecked all my solder spots to see if I didn't inadvertently connect two pins but they all look fine to me. I also resoldered the connectors for X3 and X4 but the issue persisted.

Thanks for any help or hints!

SurfGargano, 2024-11-11:
pls show a screenshot of Home and Config.

padarom, 2024-11-12:
What I just realized when comparing this controller with the working one: The working one uses version the STM32 firmware version `1.4.0_Dev_C2 build 21.08.2024`. I'm quite sure the faulty one would've had the same version, but I wanted to see if a firmware re-flash would fix the issue and the newest one I could find in the repo was 1.3.7.

Here's the screenshots you asked for:

[image]
[image]

SurfGargano, 2024-11-12:
The config looks fine.
The latest firmware is on my github : 
https://github.com/SurfGargano/VdMot_Controller/releases/latest

Funkenschuhster, 2024-12-21:
Had the same issue. Flashing no success. I bought some new from AZ Delivery. It is working fine now.

faerber93, 2025-01-11:
@Funkenschuhster , can you please post a link for the stm32 that you orderd?
I also bought some STM32 modules from AZDelivery (via Amazon) because ports 1 and 2 both seem to control port 1. With the new STM32 modules, ports 1 and 2 work correctly, but the EMF does not function – the motor doesn’t stop and ends up skipping the gears.
I will test the other STM32 modules from AZDelivery later.

Funkenschuhster, 2025-01-13:
I used these:
AZDelivery 3 x STM32F401 Development Board V 3.0 STM32F401CCU6 STM32F4 Development Board 84MHz 64KB SRAM 256KB https://amzn.eu/d/2I6qkih

Check the current factors. If you reset the controllers during startup, the factor may changed to 3.7